dioannidis/usbasp #1: bin files current?

dwillmore, 2021-05-06:
The readme.md mentions prebuilt bin files are in bin, but those files seem to be from the original commit and not updated recently.

Are they current?

dioannidis, 2021-05-06:
I'm not sure but I rebuild them anyway.

Could you please check if they are working for you ?

dwillmore, 2021-05-06:
None of the .hex files match the hash of the one I generated. Am I comparing it wrong?  Which of the files in bin/firmware should match the default main.hex from /firmware?

dioannidis, 2021-05-09:
Hi,

> Looking at the commit, you changed the .hex files in firmware/, not in bin/firmware/ Okay. Those don't match my local build, either. Compiler version differences, maybe?

  I'm using Microchip's ( Atmel ) current tool chain ( gcc version 5.4.0 (AVR_8_bit_GNU_Toolchain_3.6.2_1778) ). The firmware's in bin directory I assume that are builded from the original Fischl's code.

  The updated firmware's 1.08 are in firmware directory, main.hex for the atmega8 and m88.hex for the atmega88.

regards,

closed 2021-10-01 by dioannidis