dioannidis/usbasp #2: Build issue on Fedora 32

dwillmore, 2021-05-06:
I'm attempting to build from source on a Fedora 32 system (64 bit).  

For this task, I have installed:
avr-binutils-2.35-3.fc32.x86_64.rpm
avr-gcc-10.2.0-1.fc32.x86_64.rpm
avr-libc-2.0.0-10.fc32.noarch

cd into the firmware directory and type "make main.hex" results in most files compiling to object files.  The link step errors out with:
> avr-gcc -Wall -Wextra -Os -Iusbdrv -I. -mmcu=atmega8  -o main.bin usbdrv/usbdrv.o usbdrv/usbdrvasm.o usbdrv/oddebug.o isp.o clock.o tpi.o main.o -Wl,-Map,main.map
> /usr/lib/gcc/avr/10.2.0/../../../../avr/bin/ld: main.o:(.bss+0x1): multiple definition of `ispTransmit'; isp.o:(.bss+0x2): first defined here
> collect2: error: ld returned 1 exit status
> make: *** [Makefile:61: main.bin] Error 1

ispTransmit seems to be a function pointer and I don't see anywhere in the code where it's defined, but two places where it's assigned.

The short solution I performed was to remove the definition of the variable from isp.h and move it to isp.c.  I also added a copy to main.c, but marked that one as 'extern'. At that point everything compiled smoothly and resulted in a binary.  I'm not sure if that's how you want this fixed, so I didn't make a pull request.

dioannidis, 2021-05-06:
Hi,
 
  I missed a commit from upstream . 

  Could you please check if it is working for you ?

dwillmore, 2021-05-06:
Just pulled v1.07-54-g6337918 and it failed.

dioannidis, 2021-10-01:
Closed for no feedback

closed 2021-10-01 by dioannidis