dioannidis/usbasp #4: Found no UART support in sources

wigsoft, 2022-01-11:
First of all v1.08 is a great update on USBasp firmware!!!
As I use a little different hardware than fischls board, I had to change the led ports, clock, pins of USB, Interrupt, ... and so I had to compile the project from the source. As expected the bcdDevice is 1.08 and the WinUSB support works perfect but I cannot find any hint to uart support in the sources published (May, 6th 2021) on https://github.com/dioannidis/usbasp.  
On  https://electronoobs.io/project/491#  you told us that the sources already include WinUSB AND UART support. 
If I'm looking at the wrong places, please give me any hint where I can find the described sources (with WinUSB and UART support) of this great project.

dioannidis, 2022-01-17:
Hi,

  thank you for your feedback and sorry for the delayed answer ( Github didn't send me notification regarding this. )

  I didn't push the UART support but I don't remember why. I'll take a look this weekend.

  FYI, it's not a CDC implementation. The UART communication uses USB Control Transfers.

wigsoft, 2022-01-17:
Hello Dimitrios!
First, thanks for your answer!!!
I used your release of May 6, 2021 which is version 1.08, adapted this version to my hardware and everything works fine, also with the perfect implementation of WCID/WinUSB.
This should be the basis for my implementation of UART support. Now I read on https://electronoobs.io/project/491# that you have already done this job based on your 1.08 WCID firmware.
I would be very satisfied with both goals in a firmware from your development, since I have seen your sources and I am sure that your implementation is more professional than mine.
One question: Is there a downside to using USB Control Transfers and not the CDC implementation (you posted that fact separately)?
Thanks again and glad to find your firmware on git (perhaps with a binary of your terminal too?)

Best regards

Gerhard Windischbauer

 

+43 664 1328056

***@***.***

 

 

From: Dimitrios Chr. Ioannidis 
Sent: Monday, January 17, 2022 10:37 AM
To: dioannidis/usbasp 
Cc: wigsoft ; Author 
Subject: Re: [dioannidis/usbasp] Found no UART support in sources (Issue #4)

Hi,

thank you for your feedback and sorry for the delayed answer ( Github didn't send me notification regarding this. )

I didn't push the UART support but I don't remember why. I'll take a look this weekend.

FYI, it's not a CDC implementation. The UART communication uses USB Control Transfers.

dioannidis, 2022-01-18:
I didn't write the improved usbasp code. This repo is a fork of [nerdralph's work.](https://github.com/nerdralph/usbasp). I used it as base to add WCID support which is accepted by nerdralph and include it at his repo.

In USB 1.1 low speed ( 1.5Mbits/s ), bulk transfers, which are what CDC are based on, are not supported. The control transfers, that I used for the serial data, are using best effort delivery, which means that other USB packets ( interrupt transfers, bulk transfers ) will have higher priority, resulting on loosing serial data.

I think that I was started to changing it to interrupt transfers, but work and life caught me and I didn't finished it. 

I'll check this weekend in what state the UART code is.

dioannidis, 2022-01-31:
Well, the uart code needs some work. I estimate in a week or two I'll have something. 

Anyway, I upload the changes needed for Windows to see the USBasp as a composite device with a winusb and a hid interface. At least in my Windows 10 machine, that is.

dioannidis, 2022-01-31:
Forgot to mention that the changes are in the uart branch.

dioannidis, 2022-02-02:
I uploaded a windows avrdude builded with winusb support using cygwin64. You can find it at bin/avrdude-winusb directory.

dioannidis, 2022-02-09:
@wigsoft 

  HID UART support added. Working reliable at 9600 Baud, at least for me.

closed 2022-02-10 by dioannidis

wigsoft, 2022-02-10:
Hello Dimitrios!
I just read that you finished the UART integration project and it works with 9600 which is more than perfect for debugging! Thank you for your effort in completing this very useful project. Unfortunately I'm too stupid to find the new sources. I don't see the new files under the "old" link (https://github.com/dioannidis/usbasp)?
Please help me again to enjoy the new features.

Kind regards,
Gerhard Windischbauer

 

+43 664 1328056

***@***.***

dioannidis, 2022-02-10:
The changes are in the uart branch ( https://github.com/dioannidis/usbasp/tree/uart ) . 

I need to find time to finish the cli / library though and fix the activation / deactivation of uart .  See #5 and #6 .

wigsoft, 2022-02-10:
Thank you very much again, will immediately start to adapt the firmware to my hardware!

Freundliche Grüße

Gerhard Windischbauer

 

+43 664 1328056

***@***.***

 

From: Dimitrios Chr. Ioannidis 
Sent: Thursday, February 10, 2022 12:27 PM
To: dioannidis/usbasp 
Cc: wigsoft ; Mention 
Subject: Re: [dioannidis/usbasp] Found no UART support in sources (Issue #4)

The changes are in the uart branch ( https://github.com/dioannidis/usbasp/tree/uart ) .

I need to find time to finish the cli / library though and fix the activation / deactivation of uart . See #5 and #6 .

dioannidis, 2022-02-11:
@wigsoft 

FYI, there was a bug in the previous commit that prevended usbasp from working . Now it is fixed.

dioannidis, 2022-02-11:
@wigsoft 

Also I merged the uart to master.

wigsoft, 2022-02-16:
Hello Dimitrios,
today I could finish my adaption and your great fw runs perfect also on my hardware.
Thank you very much for that!!!

One last question: Do you provide a windows terminal binary which is compatible to your last fw-release 1.09?
If yes, please let me know where I can get this – so I can test/use the UART debugging also.

Greetings,

Gerhard Windischbauer

 

+43 664 1328056

***@***.***

dioannidis, 2022-02-16:
@wigsoft 

Regarding terminal I'll have something this weekend or so. You could monitor the #6 issue. 

Also FYI the work is not finished. There are some problems still.

dioannidis, 2022-02-28:
@wigsoft 

Just added a cli test app for windows.
Commit https://github.com/dioannidis/usbasp/commit/d24b073cf76c7ad748c84ed514e953936925aa1e 

Look at the directory bin/hiduart test/ .