dioannidis/usbasp #8: UART USBasp

JucaTec, 2022-02-28:
Hey. 

I would like to thank you for the work.

 I don't know if I'm doing something wrong but I couldn't make the UART work through the USBasp Terminal. 

Is he already active? 

Can you tell me how to send instructions to configure and an example? 

I don't have much knowledge on this subject. 

Thank you very much.

dioannidis, 2022-02-28:
There was a [build problem of the utility](https://github.com/dioannidis/usbasp/commit/86618f8af13a06381c6a70201638375ab06b6144) . Please make sure that you have the latest master. 

 If you execute USBaspHIDUART.exe -h you'll get this help message :

```
USBasp HIDUART Test App

-l List USBasp HID devices
-i Select USBasp index ( default 0 )
-b Set Baud ( default 9600 )
-r Continuous read input
-w Interactive send output

examples

Read from USBasp at index 0 with 4800 baud
USBaspHIDUART -b 4800 -r

Interactive write to USBasp at index 1 with 9600 baud
USBaspHIDUART -i 1 -w 
```

dioannidis, 2022-02-28:
README updated with the USBaspHIDUART informations.

dioannidis, 2022-02-28:
Also did you checked the connections ? See [schematic](https://github.com/dioannidis/usbasp/blob/22adfb35c24dd47e57c969438e5ef6f259662e3e/circuit/USBasp_schematics.pdf) for the TXD, RXD at the ICSP 10-pin connector.

JucaTec, 2022-02-28:
The version is the latest master.

USBaspHIDUART.exe is not opening. I don't know if it's a compatibility issue or need something additional.

I'm running on Windows 11.

Thanks for all the help.

JucaTec, 2022-02-28:
I'm sorry. By CMD and it works.

Is it possible to use the USBasp Terminal you made? This one didn't work.

dioannidis, 2022-02-28:
> USBaspHIDUART.exe is not opening. I don't know if it's a compatibility issue or need something additional.
> 
> I'm running on Windows 11.

Didn't test it on Windows 11 only on Windows 10 .

I'll see what I can do for Windows 11.

dioannidis, 2022-02-28:
> Is it possible to use the USBasp Terminal you made? This one didn't work.

Currently the USBasp Terminal doesn't support this firmware. I started to implement it but because I don't have much free  time it'll take a while. That's why I uploaded the console utility .

dioannidis, 2022-02-28:
> From the Windows 11 CMD it worked, it's already great.

Good !

> If you need testing, I'm happy to test.

Thanks.

closed 2022-02-28 by dioannidis