dioannidis/usbasp #9: Missing avrdude 7.0 with winusb support

Bibbbi, 2022-05-14:
Hi @dioannidis,

A new release from AVRDUDE v7.0 for Windows was published.
[https://github.com/mariusgreuel/avrdude/releases/tag/v7.0-windows](url)

Are you planning an update for avrdude 7.0 with winusb support?

dioannidis, 2022-05-17:
Ho @Bibbbi ,

 I missed the move from savannah to github entirely ( too busy ) ... Sorry ...

 And just saw that they change the avrdude build framework to CMake ( and MSVC for Windows ) . I'll give it a try this weekend but I remember reading in the mail lists that was a plan to add winusb . Did you try it and failed ?

dioannidis, 2022-05-17:
By the way, as I'll prefer to switch to use the official avrdude I opened an issue there https://github.com/avrdudes/avrdude/issues/968#issue-1239006912 .

dioannidis, 2022-05-18:
@Bibbbi  for now you can use the avrdude v7 from this https://github.com/mcuee/avrdude/releases/tag/v7.0 .

dioannidis, 2022-05-18:
@Bibbbi  I changed the avrdude section in the readme adding the avrdude's v7 lack of composite devices support and a link to the mcuee unofficial build .

closed 2022-05-18 by dioannidis