dioannidis/usbasp #11: Firmware does not work on some USBISP clone with ATmega88

mcuee, 2022-07-20:
From here:
* https://github.com/avrdudes/avrdude/issues/992

I have three "USB ISP Version 2.0" devices (marked "MX-U2BLSP-V5.00" on the PCB). I think the marking on the PCB is a typo, they mean to say "MX-USBISP-V5.00".

Only one of them works with the USB Composite FW here. But they all work with the FW here.
https://github.com/MCUdude/usbisp

dioannidis, 2022-07-20:
All three devices have the same fuses ? 
If not, is it possible to take a hi res picture of the one who works and one that don't work ? Maybe different PCB ?

mcuee, 2022-07-20:
I do not think it is the same as https://github.com/aleh/usbisp/issues/6#issuecomment-719816915 because your code has already leave Port D as input.

mcuee, 2022-07-20:
> All three devices have the same fuses ? If not, is it possible to take a hi res picture of the one who works and one that don't work ? Maybe different PCB ?

All three clones have the same PCB but not so sure about the component values.

dioannidis, 2022-07-20:
@mcuee 

The usbisp devices that don't work, what are their behavior ? 

Do they enumerate or the ISP programming doesn't work ?  If it's the latter could you please try the UART feature if they are enumerate ?

mcuee, 2022-07-20:
The failed device does not enumerate.

dioannidis, 2022-07-20:
Unfortunately I don't have a similar device to test . 

Just ordered some from [here](https://www.aliexpress.com/item/33058820632.html?pdp_ext_f=%7B%22sku_id%22:%2267542394862%22,%22ship_from%22:%22%22%7D&gps-id=pcStoreJustForYou&scm=1007.23125.137358.0&scm_id=1007.23125.137358.0&scm-url=1007.23125.137358.0&pvid=bd97397b-ec66-4c6b-97ac-c993661516b9&spm=a2g0o.store_pc_home.smartJustForYou_316197325.1) but they'll take some time to arrive .

mcuee, 2022-07-21:
Thanks. Take your time. Anyway, I have alternative working FW for those two units.

mcuee, 2022-07-21:
BTW, there is a version 6 PCB using ATmega88p.  You may want to add support for that one as well.
* https://github.com/aleh/usbisp/issues/8

mcuee, 2022-07-21:
BTW, you can build the FW using MSYS2 mingw32/64 avr-gcc 8.4.0 as well.

```
$ make main.hex
avr-gcc -Wall -Wextra -Os -Iusbdrv -I. -mmcu=atmega8 -DF_CPU=12000000  -c usbdrv/usbdrv.c -o usbdrv/usbdrv.o
avr-gcc -Wall -Wextra -Os -Iusbdrv -I. -mmcu=atmega8 -DF_CPU=12000000  -x assembler-with-cpp -c usbdrv/usbdrvasm.S -o usbdrv/usbdrvasm.o
avr-gcc -Wall -Wextra -Os -Iusbdrv -I. -mmcu=atmega8 -DF_CPU=12000000  -c usbdrv/oddebug.c -o usbdrv/oddebug.o
avr-gcc -Wall -Wextra -Os -Iusbdrv -I. -mmcu=atmega8 -DF_CPU=12000000  -c isp.c -o isp.o
avr-gcc -Wall -Wextra -Os -Iusbdrv -I. -mmcu=atmega8 -DF_CPU=12000000  -c clock.c -o clock.o
avr-gcc -Wall -Wextra -Os -Iusbdrv -I. -mmcu=atmega8 -DF_CPU=12000000  -x assembler-with-cpp -c tpi.S -o tpi.o
avr-gcc -Wall -Wextra -Os -Iusbdrv -I. -mmcu=atmega8 -DF_CPU=12000000  -c main.c -o main.o
avr-gcc -Wall -Wextra -Os -Iusbdrv -I. -mmcu=atmega8 -DF_CPU=12000000  -c uart.c -o uart.o
avr-gcc -Wall -Wextra -Os -Iusbdrv -I. -mmcu=atmega8 -DF_CPU=12000000  -o main.bin usbdrv/usbdrv.o usbdrv/usbdrvasm.o usbdrv/oddebug.o isp.o clock.o tpi.o main.o uart.o -Wl,-Map,main.map
rm -f main.hex main.eep.hex
avr-objcopy -j .text -j .data -O ihex main.bin main.hex

$ avr-gcc -v
Using built-in specs.
Reading specs from c:/msys64/mingw64/bin/../lib/gcc/avr/8.4.0/device-specs/specs-avr2
COLLECT_GCC=C:\msys64\mingw64\bin\avr-gcc.exe
COLLECT_LTO_WRAPPER=c:/msys64/mingw64/bin/../libexec/gcc/avr/8.4.0/lto-wrapper.exe
Target: avr
Configured with: ../configure --build=x86_64-w64-mingw32 --prefix=/mingw64 --target=avr --enable-languages=c,c++ --disable-nls --disable-libssp --disable-shared --disable-threads --disable-libgomp --disable-libada --with-dwarf2 --enable-mingw-wildcard
Thread model: single
gcc version 8.4.0 (GCC)

```

mcuee, 2022-07-21:
Today I found another cheap ATmega8A based clone (no 3.3V/5V selection, less resistors, no jumper for FW upgrade -- I have to use tweezer to manually short the chip reset pins to the ISP connector RESET pin) which does not work with the FW here, but again, the 
* https://github.com/nerdralph/usbasp

dioannidis, 2022-07-21:
I'm thinking that the MS OS 2.0 Descriptors combined with enabling long transfers may cause problems . 

BTW, all my tests were done using a atmega88pa . I tried to find / source atmega88 ICs ( DIP, TQFP ) but except ebay and aliexpress, all the known ( to me ) distributor's stock are zero ....

dioannidis, 2022-07-21:
@mcuee
Is it possible to use wireshark ( or something similar ) to log the enumeration process ?

mcuee, 2022-07-21:
> @mcuee Is it possible to use wireshark ( or something similar ) to log the enumeration process ?

This is more difficult. Let me see what I can do (under Windows 11 or Ubuntu 20.04).

mcuee, 2022-07-21:
@dioannidis 
FYI one of the two m88 USBISP clone devices now work under Linux, but both are not wokring under Windows.

dioannidis, 2022-07-21:
> In summary, I am no longer sure if this is a bug or not. It could be a false alarm for the ATmega88 USBISP v5.0 clones.

> As for the minimum ATmega8A USBASP clone, I will test it again. It came with very old usbasp FW and that is why I tried to upgrade.

All the atmega8a usbasp clones ( I have five two of them doesn't have the rxd/txd pins exposed) work without a problem . Bad flashing maybe ?

dioannidis, 2022-07-21:
> Rebooting again to Windows and it now works under Windows.

Could you please "clean" the windows registry reg USBasp VID/PID keys ?

mcuee, 2022-07-22:
> Could you please "clean" the windows registry reg USBasp VID/PID keys ?
> 
> I use [usbdview](https://www.nirsoft.net/utils/usb_devices_view.html) in administrator mode to delete the usbasp's usb registry keys during development ....

Thanks. I do that as well, with various utilities. I clean up USB devices and COM ports periodically.

mcuee, 2022-07-22:
> In summary, I am no longer sure if this is a bug or not. It could be a false alarm for the ATmega88 USBISP v5.0 clones.

@dioannidis 
I can confirm that this is a false alarm. I use a tweezer to short the two pads and it works fine to flash the latest m88.hex file (from 16-July-2022) and it works just fine. Looks like previously I have two issues.
1) I may have used the old firmware: https://github.com/dioannidis/usbasp/blob/master/bin/firmware/usbasp.atmega88.2011-05-28.hex
2) using cables to short the two pads are not stable.

```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git.exe -c usbasp -p m88 -U .\hex\usbasp_m88_16Jul2022.hex

avrdude_git.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude_git.exe: Device signature = 0x1e930a (probably m88)
avrdude_git.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
                 To disable this feature, specify the -D option.
avrdude_git.exe: erasing chip
avrdude_git.exe: reading input file ".\hex\usbasp_m88_16Jul2022.hex"
avrdude_git.exe: input file .\hex\usbasp_m88_16Jul2022.hex auto detected as Intel Hex
avrdude_git.exe: writing flash (5736 bytes):

Writing | ################################################## | 100% 2.19s

avrdude_git.exe: 5736 bytes of flash written
avrdude_git.exe: verifying flash memory against .\hex\usbasp_m88_16Jul2022.hex:
avrdude_git.exe: input file .\hex\usbasp_m88_16Jul2022.hex auto detected as Intel Hex

Reading | ################################################## | 100% 1.26s

avrdude_git.exe: 5736 bytes of flash verified

avrdude_git.exe done.  Thank you.
```

mcuee, 2022-07-22:
> > As for the minimum ATmega8A USBASP clone, I will test it again. It came with very old usbasp FW and that is why I tried to upgrade.
> 
> All the atmega8a usbasp clones ( I have five, two of them doesn't have the rxd/txd pins exposed) work without a problem . Bad flashing maybe ?

You are right. It is a bad flashing. Using the tweezer properly fixed the issue.

mcuee, 2022-07-22:
So I will close this issue as the last ATmega88 based USBISP v5.0 clone seems to have HW issues now -- I can not flash any firmware any more (last one is the 28-Feb-2022 usb composite FW).

closed 2022-07-22 by mcuee

mcuee, 2022-07-22:
BTW, for the two usbasp clones with ATmega8A and UART lines extended to the 10pin ISP connectors, the HID UART function seems to work fine (just a simple loop-back test).

```
C:\work\avr\usbasp\bin\hiduart test\windows-x86_64 [master ≡ +10 ~0 -0 !]> .\USBaspHIDUART.exe -i 0 -w

Enumerating USBasp HIDUART (Hid Api Library Version : 0.12.0 )

USBasp device configuration
---------------------------
Crystal : 12000000
Baud    : 9600

1
2
3
4
a
b
c
d

C:\work\avr\usbasp\bin\hiduart test\windows-x86_64 [master ≡ +10 ~0 -0 !]> .\USBaspHIDUART.exe -i 1 -r

Enumerating USBasp HIDUART (Hid Api Library Version : 0.12.0 )

USBasp device configuration
---------------------------
Crystal : 12000000
Baud    : 9600

1234abcd
```

mcuee, 2022-07-22:
@dioannidis 
Sorry for the false alarm.

dioannidis, 2022-07-22:
@mcuee

If I don't ask too much, is it possible to test the [libUSBUARTTerminal](https://github.com/dioannidis/libUSBUARTTerminal) ( which is a GUI HID UART Terminal ) when you have some spare time ?

I'll appreciate the feedback .

mcuee, 2022-07-23:
> If I don't ask too much, is it possible to test the [libUSBUARTTerminal](https://github.com/dioannidis/libUSBUARTTerminal) ( which is a GUI HID UART Terminal ) when you have some spare time ?
> 
> I'll appreciate the feedback .

No problem. It seems to work rather well. I am again using two usbasp loop-back for the testing.

A few potential improvement ideas.
1. Other than serial number, maybe also provide the index number (0,1,etc) in the status bar. This makes it a bit easier to differentiate the two units. Take note the following two reasons.
1) This is because that the default serial number is the same if the user just get the binary or build from unmodified source codes. This is actually not good as the USB standard requires unique serial number if the device has serial number (older Windows version may crash). So I just changed the serial number to 12345 and then it is easier. Maybe it is also good to have this item listed in the usbasp firmware readme as well.
2) The Firmware also allows to build without serial number. 

2. The label `lblMonitor` should be removed.

3. About now shows nothing. Probably a simple program verson and copyright notice should be there.

dioannidis, 2022-07-23:
> No problem. It seems to work rather well. I am again using two usbasp loop-back for the testing.

I'm happy to hear that .

>     3. The `About` menu item now shows nothing. Probably a simple program verson and copyright notice should be there.

Yeap this already fixed .

dioannidis, 2022-07-23:
> A few potential improvement ideas.
> 
>     1. Other than serial number, maybe also provide the index number (0,1,etc) in the status bar. This makes it a bit easier to differentiate the two units. Take note the following two reasons.
> 
> 
>     * This is because that the default serial number is the same if the user just get the binary or build from unmodified source codes. This is actually not good as the USB standard requires unique serial number if the device has serial number (older Windows version may crash). So I just changed the serial number to 12345 and then it is easier. Maybe it is also good to have this item listed in the usbasp firmware readme as well.
> 
>     * The Firmware also allows to build without serial number.
 
AFAIU, the issue is with the container id . The parent device needs a container id on device level to "bind" all the functions together . I'm thinking to make a utility for patching serial AND container id ( which is a guid / uuid in fact ) of the hex file, before flashing it, for those who can't build the firmware from source .
 
>     2. The label `lblMonitor` should be removed.

Currently, if, during uart is opened, a ISP read / write command issued, then the uart closes as to not interfere with ISP. 

I wanted the uart to return to open state, if it was open, after the ISP commands finished. So, I've implemented ( not uploaded yet ) one more hid interface for monitoring, which report the device state . With this information I can reopen the uart if it was opened before the ISP commands started.

mcuee, 2022-07-24:
> AFAIU, the issue is with the container id . The parent device, in composite devices, needs a container id on device level to "bind" all the functions together . I'm thinking to make a utility for patching serial AND container id ( which is a guid / uuid in fact ) of the hex file, before flashing it, for those who can't build the firmware from source .

This will be a good improvement.

Not so sure if you can add a feature report to update this in run-time (I have PICKit 2 which does that with a -n command).

mcuee, 2022-07-24:
> Currently, if, during uart is opened, a ISP read / write command issued, then the uart closes as to not interfere with ISP.
> 
> I wanted the uart to return to ~open state~ its previous state ( to closed if it was closed or to open if it was opened ), after the ISP commands finished. So, I've implemented ( not uploaded yet ) one more hid interface for monitoring, which report the device state . With this information I can reopen the uart if it was opened before the ISP commands started.
> 
> The label lblMonitor is the current USBasp state, the extra hid interface reports.

This is a nice improvement as well. Thanks.

dioannidis, 2022-07-24:
@mcuee 

https://github.com/dioannidis/libUSBUARTTerminal/releases/tag/v0.7.1

Thx for your feedback