dioannidis/usbasp #13: UART retaining state after ISP commands finished

dioannidis, 2022-07-24:
> Currently, if, during uart is opened, a ISP read / write command issued, then the uart closes as to not interfere with ISP.
> 
> I wanted the uart to return to ~open state~ its previous state ( to closed if it was closed or to open if it was opened ), after the ISP commands finished. So, I've implemented ( not uploaded yet ) one more hid interface for monitoring, which report the device state . With this information I can reopen the uart if it was opened before the ISP commands started.
> 
> The label lblMonitor is the current USBasp state, the extra hid interface reports.

This is a nice improvement as well. Thanks.

_Originally posted by @mcuee in https://github.com/dioannidis/usbasp/issues/11#issuecomment-1193231887_

dioannidis, 2022-07-25:
@mcuee

My tests regarding this are successful. The [libUSBUARTTerminal](https://github.com/dioannidis/libUSBUARTTerminal) already has support for this functionality. I need to add it to USBaspHIDUART utility also.

If you're going to test this I'll wait otherwise I'll closed it as resolved.

mcuee, 2022-07-25:
You can close it for now. If I encountered issues later, I will report it here.

closed 2022-07-25 by dioannidis