dioannidis/usbasp #18: @llinjupt

dioannidis, 2023-03-30:
@dioannidis
Just wonder if you are willing to add the support in your usbasp firmware. You need to update the following code as per the suggestion from @llinjupt 
https://github.com/dioannidis/usbasp/blob/master/firmware/isp.c

_Originally posted by @mcuee in https://github.com/avrdudes/avrdude/issues/1304#issuecomment-1489615806_

dioannidis, 2023-04-25:
@mcuee, @llinjupt 

If you have a AT89S51 IC or a AT89S52 IC is possible to the reversed polarity probe in this usbasp firmware https://nephelae.eu/Firmware_ATMega8_12MHz_TPI_HIDUART_SNWRITE_AT89.zip ?

mcuee, 2023-04-25:
Great. I will get the AT89S52 soon and I will carryout the test.

So you are using the default avrdude 7.1 MinGW build binary, right?

dioannidis, 2023-04-25:
> So you are using the default avrdude MinGW build binary without the need to change avrdude, right?

I used this version avrdude: Version 7.1-20230420 (b038bd4) which works !

mcuee, 2023-05-07:
@dioannidis 

Thanks. I just got AT89S51 chip. The first test shows it works well. Great work.

I use the 1.11 pre-release FW (https://github.com/dioannidis/usbasp/releases/download/v1.11/Firmware_ATMega8_12MHz_TPI_HIDUART_SNWRITE.zip).

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p at89s51
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e5106 (probably 89S51)

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p at89s51 -U .\entest.hex
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e5106 (probably 89S51)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\entest.hex for flash
         with 512 bytes in 1 section within [0, 0x1ff]
avrdude: writing 512 bytes flash ...
Writing | ################################################## | 100% 1.32 s
avrdude: 512 bytes of flash written
avrdude: verifying flash memory against .\entest.hex
Reading | ################################################## | 100% 0.61 s
avrdude: 512 bytes of flash verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p at89s51 -t
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e5106 (probably 89S51)
avrdude> dump flash
Reading | ################################################## | 100% 0.45 s
0000  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |
0010  66 6f 78 20 6a 75 6d 70  73 20 6f 76 65 72 20 74  |fox jumps over t|
0020  68 65 20 6c 61 7a 79 20  64 6f 67 0a 54 68 65 20  |he lazy dog The |
0030  71 75 69 63 6b 20 62 72  6f 77 6e 20 66 6f 78 20  |quick brown fox |
0040  6a 75 6d 70 73 20 6f 76  65 72 20 74 68 65 20 6c  |jumps over the l|
0050  61 7a 79 20 64 6f 67 0a  54 68 65 20 71 75 69 63  |azy dog The quic|
0060  6b 20 62 72 6f 77 6e 20  66 6f 78 20 6a 75 6d 70  |k brown fox jump|
0070  73 20 6f 76 65 72 20 74  68 65 20 6c 61 7a 79 20  |s over the lazy |
0080  64 6f 67 0a 54 68 65 20  71 75 69 63 6b 20 62 72  |dog The quick br|
0090  6f 77 6e 20 66 6f 78 20  6a 75 6d 70 73 20 6f 76  |own fox jumps ov|
00a0  65 72 20 74 68 65 20 6c  61 7a 79 20 64 6f 67 0a  |er the lazy dog |
00b0  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |
00c0  66 6f 78 20 6a 75 6d 70  73 20 6f 76 65 72 20 74  |fox jumps over t|
00d0  68 65 20 6c 61 7a 79 20  64 6f 67 0a 54 68 65 20  |he lazy dog The |
00e0  71 75 69 63 6b 20 62 72  6f 77 6e 20 66 6f 78 20  |quick brown fox |
00f0  6a 75 6d 70 73 20 6f 76  65 72 20 74 68 65 20 6c  |jumps over the l|
avrdude> dump flash
Reading | ################################################## | 100% 0.45 s
0100  61 7a 79 20 64 6f 67 0a  54 68 65 20 71 75 69 63  |azy dog The quic|
0110  6b 20 62 72 6f 77 6e 20  66 6f 78 20 6a 75 6d 70  |k brown fox jump|
0120  73 20 6f 76 65 72 20 74  68 65 20 6c 61 7a 79 20  |s over the lazy |
0130  64 6f 67 0a 54 68 65 20  71 75 69 63 6b 20 62 72  |dog The quick br|
0140  6f 77 6e 20 66 6f 78 20  6a 75 6d 70 73 20 6f 76  |own fox jumps ov|
0150  65 72 20 74 68 65 20 6c  61 7a 79 20 64 6f 67 0a  |er the lazy dog |
0160  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |
0170  66 6f 78 20 6a 75 6d 70  73 20 6f 76 65 72 20 74  |fox jumps over t|
0180  68 65 20 6c 61 7a 79 20  64 6f 67 0a 54 68 65 20  |he lazy dog The |
0190  71 75 69 63 6b 20 62 72  6f 77 6e 20 66 6f 78 20  |quick brown fox |
01a0  6a 75 6d 70 73 20 6f 76  65 72 20 74 68 65 20 6c  |jumps over the l|
01b0  61 7a 79 20 64 6f 67 0a  54 68 65 20 71 75 69 63  |azy dog The quic|
01c0  6b 20 62 72 6f 77 6e 20  66 6f 78 20 6a 75 6d 70  |k brown fox jump|
01d0  73 20 6f 76 65 72 20 74  68 65 20 6c 61 7a 79 20  |s over the lazy |
01e0  64 6f 67 0a 54 68 65 20  71 75 69 63 6b 20 62 72  |dog The quick br|
01f0  6f 77 6e 20 66 6f 78 20  6a 75 6d 70 73 20 6f 76  |own fox jumps ov|
avrdude> quit

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p at89s51 -t
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e5106 (probably 89S51)
avrdude> part
AVR Part                      : AT89S51
Chip Erase delay              : 250000 us
RESET disposition             : dedicated
RETRY pulse                   : SCK
Serial program mode           : yes
Parallel program mode         : yes
Timeout                       : 200
StabDelay                     : 100
CmdexeDelay                   : 25
SyncLoops                     : 32
PollIndex                     : 4
PollValue                     : 0x69
Memory Detail                 :

                                  Block Poll               Page                       Polled
  Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
  ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
  flash                   2    10   256    0 no       4096    1      0  4500  4500 0xff 0xff
  lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
  signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
avrdude> dump lock
Reading | ################################################## | 100% 0.00 s
0000  00                                                |.               |
avrdude> dump signature
Reading | ################################################## | 100% 0.01 s
0000  1e 51 06                                          |.Q.             |
avrdude> quit

avrdude done.  Thank you.
```

dioannidis, 2023-05-07:
Hi,

@mcuee 

I'll wait for @orbanp and / or @llinjupt to test the fw also, if they can. Then I'll make the release.

Thank you very much for the testing and the feedback !

regards,

closed 2023-05-07 by dioannidis