dioannidis/usbasp #21: clock problem ?

Franck78, 2023-05-06:
Hi
I flashed v1.11  _USBasp_ATMega8_12MHz_TPI_HIDUART_SNWRITE.hex_
succesfully, with fuses and the result is

```
mai 06 17:31:30  kernel: usb 2-1.5: new full-speed USB device number 16 using ehci-pci
mai 06 17:31:30  kernel: usb 2-1.5: device descriptor read/64, error -32
mai 06 17:31:30  kernel: usb 2-1.5: device descriptor read/64, error -32

after some trial with fuses, back to fisch 

mai 06 19:19:52 : usb 2-1.5: new low-speed USB device number 51 using ehci-pci
mai 06 19:19:52 : usb 2-1.5: New USB device found, idVendor=16c0, idProduct=05dc, bcdDevice= 1.05
mai 06 19:19:52 : usb 2-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
mai 06 19:19:52 : usb 2-1.5: Product: USBasp
.....
```
Common causes says some clocking problem here.
The device is the common thing from aliexpress, based on atmega8.

Note:
it is an absolute waste of time to make the user 'build' the final binary to flash the atmega. 

On the raspberry I used to flash , I did not go beyond  _apt-get install gcc-avc_ when compiler started complaining about _fatal error: avr/io.h:_
Just no point. Not the goal.

Simply try to put result of your compiler in 'bin/firmware', with explicit names for variants mega8 maga88.
Remove confusing crap (2007 ?? 2009 ??? 2011???)

keep fitsch.hex with the correct name (v1.5.hex= no;  v1.05.hex = yes)

and try to find the USB problem ;)

Thank you

dioannidis, 2023-05-06:
Hi,

  I assume that you used the firmware from the [1.11 pre-release](https://github.com/dioannidis/usbasp/releases/tag/v1.11) not the file from the URL I posted in another message.

> mai 06 17:31:30  kernel: usb 2-1.5: **new full-speed USB device** number 16 using ehci-pci
> mai 06 17:31:30  kernel: usb 2-1.5: device descriptor read/64, error -32
> mai 06 17:31:30  kernel: usb 2-1.5: device descriptor read/64, error -32

The logs above shows that you connected probably a full speed USB device. AFAIK, the chinese USBasp clones ( if not all then the 98% )  are low speed devices . The speed identification of a USB device, being low speed or full speed, [is indicated from the hardware not by the firmware](https://www.beyondlogic.org/usbnutshell/usb2.shtml#SpeedIdentification).

Can you post a photo of your device ?

regards,

dioannidis, 2023-05-06:
Hi,

> Note:
> it is an absolute waste of time to make the user 'build' the final binary to flash the atmega.

I just noticed that for the 1.10 release the firmware binaries are missing. I'll uploaded them over the next few days. Sorry for that.

regards,

Franck78, 2023-05-06:
I flashed the hex you posted a week ago  under v1.11 pre release. ATmega8, 12MHz
Because it was built.

Photo of the cheap aliexpress
[image]

dioannidis, 2023-05-06:
Hi,

  confirmed that the binaries also in the 1.11 pre release are corrupted ! Re uploading them shortly ! 

regards,

dioannidis, 2023-05-06:
Hi,
  
  can you please test again with the new firmware binary ?

regards,

Franck78, 2023-05-06:
```
mai 06 22:26:40  : usb 2-1.5: new low-speed USB device number 55 using ehci-pci
mai 06 22:26:40  : usb 2-1.5: No LPM exit latency info found, disabling LPM.
mai 06 22:26:40  : usb 2-1.5: New USB device found, idVendor=16c0, idProduct=05dc, bcdDevice= 1.11
mai 06 22:26:40  : usb 2-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
mai 06 22:26:40  : usb 2-1.5: Product: USBasp
mai 06 22:26:40  : usb 2-1.5: Manufacturer: www.fischl.de
mai 06 22:26:40  : hid-generic 0003:16C0:05DC.0004: hiddev96,hidraw1: USB HID v1.01 Device [www.fischl.de USBasp] on usb-0000:00:1d.0-1.5/input1
mai 06 22:26:40  : hid-generic 0003:16C0:05DC.0005: hiddev97,hidraw2: USB HID v1.01 Device [www.fischl.de USBasp] on usb-0000:00:1d.0-1.5/input2

```

Yep, much better. 

Quick test under arduino IDE 1.88, cannot upload with programmer=USBASP

dioannidis, 2023-05-06:
Hi,

  thank you very much for your feedback !

> Quick test under arduino IDE 1.88, cannot upload with programmer=USBASP

Could you please open a new issue for the above ?

regards,

closed 2023-05-06 by dioannidis