dioannidis/usbasp #22: random read

Franck78, 2023-05-06:
With firmware 1.11

random read somewhere !

between computer and usbasp
or
between usbasp and device programmed

```
avrdude: auto set sck period (because given equals null)
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x000307
avrdude: Expected signature for ATmega8 is 1E 93 07
         Double check chip, or use -F to override this check.

avrdude done.  Thank you.

```
some read:
avrdude: Device signature = 0x010103
avrdude: Device signature = 0x013505
avrdude: Device signature = 0x00032f

Going back to fisch 1.05 and everything  is ok.

dioannidis, 2023-05-06:
Hi,

  could you plz post which os, os version and avrdude version do you have ?

regards,

dioannidis, 2023-05-06:
Hi,

I can't reproduced it in Windows 10 and in Debian 11 ( running in a VM ) .

regards,

Franck78, 2023-05-06:
avrdude: Version 6.3-20171130 (on the raspberry)
for programming the USBASP

avrdude: Version 6.1, (openSUSE 42.3 Buildservice)
for programming using the USBASP

Franck78, 2023-05-06:
that will be difficult to fix. it's a bug in reading the device I think.
'bit banging' bug.

Timing problem surely

dioannidis, 2023-05-06:
Hi,
  
  do you have the same problem if you connect the usbasp to rasp and use the 6.3 avrdude  ?

  If yes, I could try to reproduced it with my old raspberry pi 2 over the next few days .  

regards,

dioannidis, 2023-05-06:
Hi,

> If yes, I could try to reproduced it with my old raspberry pi 2 over the next few days .

  with raspberry's avrdude I cannot reproduced it either !

```
pi@raspberrypi:~ $ cat /etc/debian_version
10.12
pi@raspberrypi:~ $ avrdude -v

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/home/pi/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

avrdude: no programmer has been specified on the command line or the config file
         Specify a programmer using the -c option and try again

```

Testing 100 times using bash :

```
pi@raspberrypi:~ $ for i in {1..100}; do sudo avrdude -c usbasp -p m8; done &> usbasp_stability.dump
pi@raspberrypi:~ $ cat usbasp_stability.dump | grep Expected
pi@raspberrypi:~ $
pi@raspberrypi:~ $ cat usbasp_stability.dump | more

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9307 (probably m8)

avrdude: safemode: Fuses OK (E:FF, H:C5, L:BF)

avrdude done.  Thank you.

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9307 (probably m8)

avrdude: safemode: Fuses OK (E:FF, H:C5, L:BF)

avrdude done.  Thank you.

--------- 98 times -----------

```

regards,

dioannidis, 2023-05-06:
@mcuee

If you have avrdude 6.1 in linux, can you please test to see if you can reproduce this behavior ?

Franck78, 2023-05-07:
USBASP under fisch1.05

plugged on my PI_1, I can read one time. Next time is error. I have to unplug the usb.
With my laptop Opensuse, no need to unplug, it works fine.

USBASP under fw1.11

same random reading .
Device id changing all the the time. But code read seems similar.
And no need to unplug it...

```
root@rpi:~# avrdude -v -cusbasp -pm8

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/root/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbasp
         AVR Part                      : ATmega8
         Chip Erase delay              : 10000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53                                                                                                                           
         Memory Detail                 :                                                                                                                                
                                                                                                                                                                        
                                  Block Poll               Page                       Polled                                                                            
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack                                                                            
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------                                                                            
           eeprom         4    20   128    0 no        512    4      0  9000  9000 0xff 0xff
           flash         33    10    64    0 yes      8192   64    128  4500  4500 0xff 0x00
           lfuse          0     0     0    0 no          1    0      0  2000  2000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  2000  2000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  2000  2000 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x013707
avrdude: Expected signature for ATmega8 is 1E 93 07
         Double check chip, or use -F to override this check.

avrdude: Device signature = 0x133732
avrdude: Device signature = 0x130307
avrdude: Device signature = 0x010301
avrdude: Device signature = 0x010304
avrdude: Device signature = 0x013704
avrdude: Device signature = 0x010104
[... 122 lines cut ...]
  iProduct                2 USBasp
  iSerial                 3 (error)
  bNumConfigurations      1
```
  iSerial                 3 (error)  !!??

```
avrdude -v -cusbasp -pm8  -B10
avrdude: set SCK frequency to 93750 Hz
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9307 (probably m8)
avrdude: safemode: lfuse reads as C4
avrdude: safemode: hfuse reads as D9

avrdude: safemode: lfuse reads as C4
avrdude: safemode: hfuse reads as D9
avrdude: safemode: Fuses OK (E:FF, H:D9, L:C4)

```
reading correctly with -B1 .. B7  ... -B10

B11, random read

mcuee, 2023-05-07:
@Franck78 

Can you try latest avrdude 7.1 release or even avrdude git main? There are many fixes in avrdude 7.1 and latest git main which may help.

dioannidis, 2023-05-07:
Hi,

 @Franck78

Can you please try with this fw http://www.nephelae.eu/USBasp_ATMega8_12MHz_TPI_HIDUART_SNWRITE_fix_tm.zip ? ( no need to flash eeprom again )

@mcuee

I think I know what might be the problem ...

regards,

Franck78, 2023-05-07:
no noticeable change with . Still need -B10

I will try latter with avrdude 7.

dioannidis, 2023-05-07:
Hi,

@Franck78, @mcuee

IMHO its not a problem of the firmware or avrdude, it's the connection between the USBasp and your board / mcu which cannot communicate at 750 kHz and above for some reason . 

This fw uses automatic SCK speed negotiation and a higher default SCK clock speed. When a speed is found where the target responds, for subsequent communications with the target, the speed is doubled, since the slowest speed is only needed the first time the PE command is received after power-up.

When you used -B 10,  USBasp enters programming mode at 375 kHz and then doubles it to 750 kHz for the rest of the communications . This worked well and it's the first time I got a report that this fails . Anyway lets see if I'm wrong ...

regards,

mcuee, 2023-05-07:
@dioannidis 

I believe your analysis is good. I can not reproduce the issue under Ubuntu 20.04, with either avrdude 6.1 release or 6.3 release.

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_release/avrdude-6.1$ ./avrdude -C ./avrdude.conf -p m8

avrdude: no programmer has been specified on the command line or the config file
         Specify a programmer using the -c option and try again

mcuee@UbuntuSwift3:~/build/avr/avrdude_release/avrdude-6.1$ ./avrdude -C ./avrdude.conf -c usbasp -p m8

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9307

avrdude: safemode: Fuses OK (E:FF, H:D9, L:9F)

avrdude done.  Thank you.

mcuee@UbuntuSwift3:~/build/avr/avrdude_release/avrdude-6.1$ cd ..
mcuee@UbuntuSwift3:~/build/avr/avrdude_release$ cd avrdude-6.3/
mcuee@UbuntuSwift3:~/build/avr/avrdude_release/avrdude-6.3$ ./avrdude -C ./avrdude.conf -c usbasp -p m8

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9307 (probably m8)

avrdude: safemode: Fuses OK (E:FF, H:D9, L:9F)

avrdude done.  Thank you.
```

dioannidis, 2023-05-07:
#23

dioannidis, 2023-05-07:
Hi,

 @Franck78 

  Can you test this fw www.nephelae.eu/USBasp_ATMega8_12MHz_TPI_HIDUART_SNWRITE_2hid_fix_2.zip ?

regards,

Franck78, 2023-05-07:
Working now with 2hid_fix2 and no '-B10'

another bug/request for that ?
```
mai 07 11:34:18 : hid-generic 0003:16C0:05DC.0008: hiddev96,hidraw1: USB HID v1.01 Device [www.fischl.de USBasp] on usb-0000:00:1d.0-1.5/input1
mai 07 11:34:18 : hid-generic 0003:16C0:05DC.0009: hiddev97,hidraw2: USB HID v1.01 Device [www.fischl.de USBasp] on usb-0000:00:1d.0-1.5/input2
```
I was secretly expecting a serial device as 'ttyACM0' ;)

dioannidis, 2023-05-07:
Hi,

@Franck78, @mcuee 

> Working now with 2hid_fix2 and no '-B10'

Great ! I'll update the 1.11 pre release with the new binaries .

Thank you very much for your feedback and for testing !

> another bug/request for that ?
> I was secretly expecting a serial device as 'ttyACM0' ;)

Unfortunately, the specs for USB 1.1 for low speed devices doesn't support bulk transfers which the USB CDC needs and besides that you'll need to modify your hardware ( USBasp ) to indicate that is a full speed USB device .

You can use the [usbasphiduart utility](https://github.com/dioannidis/usbasp#usbasphiduart-utility) or you can implement the [UART HID protocol](https://github.com/dioannidis/usbasp#uart-hid-protocol) your self .

regards,

Franck78, 2023-05-07:
moving the 'R3' from 'd-' to 'd+' is not the problem on the cheap chineese usbasp. The question is:

Can the USB stack inside the atmega8 implement the full speed when hardware says 'full speed' !

dioannidis, 2023-05-07:
Hi,

> Can the USB stack inside the atmega8 implement the full speed when hardware says 'full speed' !

The [V-USB](https://www.obdev.at/products/vusb/index.html) driver which USBasp is based on doesn't support full speed .

And there are other issues i.e. Low Speed is 1.5 MHz but Full Speed is 12 MHz . The atmega8 max clock is 16 MHz and ( if you a have a USBisp ) the atmega88 max clock is 20MHz . even if you change the crystals, those speeds are not enough for sampling the usb signals . 

regards,

mcuee, 2023-05-07:
> Can the USB stack inside the atmega8 implement the full speed when hardware says 'full speed' !

As mentioned by @dioannidis, V-USB is based on FW and cannot support full-speed USB.

If you want to have proper USB to Serial support inside FW (using USB CDC-ACM), then you need to use AVR chips with native USB support, like ATmega32U4 or things like that. For example, Arduino is using ATmeag16U2 for the USB-DFU+USB-Serial in the Arduino Uno/Mega.
https://github.com/arduino/ArduinoCore-avr/tree/master/firmwares/atmegaxxu2

Franck78, 2023-05-07:
yep that was what I was thinking, with 12MHz crystal, they better be virtuose assembler writers.

All my 'confusion' is from #21 first comment: why did it appear as 'full-speed' when hardware says low-speed.
Anyway, fixed.

Franck78, 2023-05-11:
FYI:

I took a supposed brand new sealed ATMEGA8 for programming
It needed a -B7 to read correctly. Without that, same bad reading.

With -B7 I got the fuse in it
```
ller2:~> avrdude -v   -cusbasp -pm8 -U lfuse:w:0xc4:m -B7

avrdude: Version 6.1, (openSUSE Buildservice)
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/home/fbourdonnec/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbasp
         Setting bit clk period        : 7.0
         AVR Part                      : ATmega8
         Chip Erase delay              : 10000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    20   128    0 no        512    4      0  9000  9000 0xff 0xff
           flash         33    10    64    0 yes      8192   64    128  4500  4500 0xff 0x00
           lfuse          0     0     0    0 no          1    0      0  2000  2000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  2000  2000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  2000  2000 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: set SCK frequency to 93750 Hz
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9307
avrdude: safemode: lfuse reads as E1
avrdude: safemode: hfuse reads as D9
avrdude: reading input file "0xc1"
avrdude: writing lfuse (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lfuse written
avrdude: verifying lfuse memory against 0xc4:
avrdude: load data lfuse data from input file 0xc4:
avrdude: input file 0xc4 contains 1 bytes
avrdude: reading on-chip lfuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of lfuse verified

avrdude: safemode: lfuse reads as C4
avrdude: safemode: hfuse reads as D9
avrdude: safemode: Fuses OK (E:FF, H:D9, L:C4)

avrdude done.  Thank you.

```
C4 is the Lfuse that is in the chip that originaly prompted this report.

and it works

```
avrdude: auto set sck period (because given equals null)
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9307
avrdude: safemode: lfuse reads as C4
avrdude: safemode: hfuse reads as D9

avrdude: safemode: lfuse reads as C4
avrdude: safemode: hfuse reads as D9
avrdude: safemode: Fuses OK (E:FF, H:D9, L:C4)

avrdude done.  Thank you.

```

dioannidis, 2023-05-11:
Hi,

> avrdude: Device signature = 0x1e9307
> avrdude: safemode: lfuse reads as E1
> avrdude: safemode: hfuse reads as D9
> avrdude: reading input file "0xc1"

 test it an atmega8a with the above fuses and I can reproduce the issue .

regards,

closed 2023-06-04 by dioannidis

dioannidis, 2023-06-04:
@Franck78 

When you have time can you test this fix https://github.com/dioannidis/usbasp/releases/download/v1.11/Firmware_ATMega8_12MHz_TPI_HIDUART_SNWRITE_FIX_LOW_SPEED.zip ?

Tested with an atmega8 with hfuse 0xD9 and lfuse 0xE1 

regards,

dioannidis, 2023-06-05:
@Franck78 

The [v1.11](https://github.com/dioannidis/usbasp/releases/tag/v1.11) is released 

regards

commit 28398320 (2023-06-04): Disabling the doubling SCK clock speed after PE fixes the read errors . Need to investigate more . Fix #22