dioannidis/usbasp #23: Firmware version 1.11 problem with ATtiny13A

mcuee, 2023-05-07:
Interesting I have issues with ATtiny13A using the 1.11 FW (no problem with ATmega328PB, ATmega32A and ATmega8A).

I can reproduce the issue under Ubuntu Linux 20.04 and Windows 11. The ATtiny13A breakout board seems to be okay as I tested it with other programmers as well.

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./USBaspHIDUART -u 111

Enumerating USBasp HIDUART (Hid Api Library Version : 0.14.0 )
-----------
USBasp List index : 0

Type : 16C0 05DC
Path : 3-1:1.1
Serial number : 3456
Manufacturer  : www.fischl.de
Product       : USBasp
Release       : 1.11
Interface     : 1

Updating USBasp Serial Number ... Done .

Enumerating USBasp HIDUART (Hid Api Library Version : 0.14.0 )
-----------
USBasp List index : 0

Type : 16C0 05DC
Path : 3-1:1.1
Serial number : 0111
Manufacturer  : www.fischl.de
Product       : USBasp
Release       : 1.11
Interface     : 1
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -c usbasp -p t13a
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x000107
avrdude error: expected signature for ATtiny13A is 1E 90 07
        double check chip or use -F to override this check

avrdude done.  Thank you.

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -c usbasp -p t13a -B10
avrdude: set SCK frequency to 93750 Hz
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e0306
avrdude error: expected signature for ATtiny13A is 1E 90 07
        double check chip or use -F to override this check

avrdude done.  Thank you.
```

mcuee, 2023-05-07:
No issues with Firmware version 1.10.

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./USBaspHIDUART -u 110

Enumerating USBasp HIDUART (Hid Api Library Version : 0.14.0 )
-----------
USBasp List index : 0

Type : 16C0 05DC
Path : 3-1:1.1
Serial number : 8888
Manufacturer  : www.fischl.de
Product       : USBasp
Release       : 1.10
Interface     : 1

USBasp does not support Serial Number update ! 

Enumerating USBasp HIDUART (Hid Api Library Version : 0.14.0 )
-----------
USBasp List index : 0

Type : 16C0 05DC
Path : 3-1:1.1
Serial number : 8888
Manufacturer  : www.fischl.de
Product       : USBasp
Release       : 1.10
Interface     : 1
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -c usbasp -p t13a
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9007 (probably t13a)

avrdude done.  Thank you.
```

mcuee, 2023-05-07:
Full debug log for FW version 1.11, with and without `-B10`.

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -c usbasp -p t13a -B10 -vvvv

avrdude: Version 7.1-20230108
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /home/mcuee/build/avr/avrdude_bin/avrdude.conf
         User configuration file is /home/mcuee/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbasp
         Setting bit clk period        : 10.0
avrdude: usbasp_open("usb")
avrdude: seen device from vendor >www.fischl.de<
avrdude: seen product >USBasp<
         AVR Part                      : ATtiny13A
         Chip Erase delay              : 4000 us
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65     5     4    0 no         64    4      0  4000  4000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           flash                  65    10    32    0 yes      1024   32     32  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           calibration             0     0     0    0 no          2    1      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/
avrdude: usbasp_initialize()
avrdude: usbasp_transmit("USBASP_FUNC_GETCAPABILITIES", 0x00, 0x00, 0x00, 0x00)
         <= [a1] [00] [00] [00] 
avrdude: usbasp_spi_set_sck_period(1e-05)
avrdude: try to set SCK period to 1e-05 s (= 100000 Hz)
avrdude: connected USBasp is not cabable of 3 MHz SCK
[... 101 lines cut ...]
         <= [00] 
avrdude: AVR device initialized and ready to accept instructions
Reading |                                                    | 0% 0.00 s 
avrdude: usbasp_spi_cmd(0x30, 0x00, 0x00, 0x00) ...
avrdude: usbasp_transmit("USBASP_FUNC_TRANSMIT", 0x30, 0x00, 0x00, 0x00)
         <= [00] [60] [01] [01] 
avrdude: usbasp_spi_cmd() => 0x00, 0x60, 0x01, 0x01
avrdude: usbasp_spi_cmd(0x30, 0x00, 0x01, 0x00) ...
avrdude: usbasp_transmit("USBASP_FUNC_TRANSMIT", 0x30, 0x00, 0x01, 0x00)
         <= [00] [61] [00] [03] 
avrdude: usbasp_spi_cmd() => 0x00, 0x61, 0x00, 0x03
Reading | #################                                  | 33% 0.00 s 
avrdude: usbasp_spi_cmd(0x30, 0x00, 0x02, 0x00) ...
avrdude: usbasp_transmit("USBASP_FUNC_TRANSMIT", 0x30, 0x00, 0x02, 0x00)
         <= [00] [60] [01] [04] 
avrdude: usbasp_spi_cmd() => 0x00, 0x60, 0x01, 0x04
Reading | ################################################## | 100% 0.00 s 
avrdude: device signature = 0x010304
avrdude main() [main.c:1421] error: expected signature for ATtiny13A is 1E 90 07
        double check chip or use -F to override this check
avrdude: usbasp_close()
avrdude: usbasp_transmit("USBASP_FUNC_DISCONNECT", 0x00, 0x00, 0x00, 0x00)

avrdude done.  Thank you.
```

mcuee, 2023-05-07:
Run log under Windows.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\USBaspHIDUART.exe -u 0111

Enumerating USBasp HIDUART (Hid Api Library Version : 0.12.0 )
-----------
USBasp List index : 0

Type : 16C0 05DC
Path : \\?\HID#VID_16C0&PID_05DC&MI_01#7&192e7c9b&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Serial number : 0000
Manufacturer  : www.fischl.de
Product       : USBasp
Release       : 1.11
Interface     : 1

Updating USBasp Serial Number ... Done .

Enumerating USBasp HIDUART (Hid Api Library Version : 0.12.0 )
-----------
USBasp List index : 0

Type : 16C0 05DC
Path : \\?\HID#VID_16C0&PID_05DC&MI_01#7&192e7c9b&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Serial number : 0111
Manufacturer  : www.fischl.de
Product       : USBasp
Release       : 1.11
Interface     : 1
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p t13a
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0xff2105
avrdude error: expected signature for ATtiny13A is 1E 90 07
        double check chip or use -F to override this check

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p t13a -B10
avrdude: set SCK frequency to 93750 Hz
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x00ff05
avrdude error: expected signature for ATtiny13A is 1E 90 07
        double check chip or use -F to override this check

avrdude done.  Thank you.
```

dioannidis, 2023-05-07:
Hi,

  Can you test with this fw www.nephelae.eu/USBasp_ATMega8_12MHz_TPI_HIDUART_SNWRITE_2hid_fix.zip

regards,

mcuee, 2023-05-07:
> Can you test with this fw [www.nephelae.eu/USBasp_ATMega8_12MHz_TPI_HIDUART_SNWRITE_2hid_fix_2.zip](http://www.nephelae.eu/USBasp_ATMega8_12MHz_TPI_HIDUART_SNWRITE_2hid_fix_2.zip) ?

@dioannidis 

Great. This new hex file fixed the issue with ATtiny13A.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p m8a -U flash:v:.\USBasp_ATMega8_12MHz_TPI_HIDUART_SNWRITE_2hid_fix_2.hex:i
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9307 (probably m8a)
avrdude: verifying flash memory against .\USBasp_ATMega8_12MHz_TPI_HIDUART_SNWRITE_2hid_fix_2.hex
Reading | ################################################## | 100% 1.28 s
avrdude: 6070 bytes of flash verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p t13a
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9007 (probably t13a)

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p t13a
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9007 (probably t13a)

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p t13a -U .\entest.hex
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9007 (probably t13a)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\entest.hex for flash
         with 512 bytes in 1 section within [0, 0x1ff]
         using 16 pages and 0 pad bytes
avrdude: writing 512 bytes flash ...
Writing | ################################################## | 100% 0.52 s
avrdude: 512 bytes of flash written
avrdude: verifying flash memory against .\entest.hex
Reading | ################################################## | 100% 0.33 s
avrdude: 512 bytes of flash verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p t13a -B10
avrdude: set SCK frequency to 93750 Hz
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9007 (probably t13a)

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p t13a -U .\entest.hex -B10
avrdude: set SCK frequency to 93750 Hz
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9007 (probably t13a)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: set SCK frequency to 93750 Hz
avrdude: reading input file .\entest.hex for flash
         with 512 bytes in 1 section within [0, 0x1ff]
         using 16 pages and 0 pad bytes
avrdude: writing 512 bytes flash ...
Writing | ################################################## | 100% 0.53 s
avrdude: 512 bytes of flash written
avrdude: verifying flash memory against .\entest.hex
Reading | ################################################## | 100% 0.33 s
avrdude: 512 bytes of flash verified

avrdude done.  Thank you.
```

dioannidis, 2023-05-07:
Hi,

@mcuee 

When you'll have time, can you please test it with more MCUs / Boards ?

regrads,

closed 2023-05-07 by dioannidis

mcuee, 2023-05-07:
Yes, I have tested ATmega8A, ATtiny88, ATmega328P and ATmega256 and all are okay.

dioannidis, 2023-05-07:
Hi,

@mcuee, @Franck78

The binaries at 1.11 pre release are updated . I downloaded them and tested them ( flash, programming, uart and sn write ) but a second pair of eyes of testing doesn't hurt ;) .

regards,

mcuee, 2023-05-07:
I did some sanity check and the updated hex file works fine.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\USBaspHIDUART.exe -u 1119

Enumerating USBasp HIDUART (Hid Api Library Version : 0.12.0 )
-----------
USBasp List index : 0

Type : 16C0 05DC
Path : \\?\HID#VID_16C0&PID_05DC&MI_01#7&192e7c9b&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Serial number : 0000
Manufacturer  : www.fischl.de
Product       : USBasp
Release       : 1.11
Interface     : 1

Updating USBasp Serial Number ... Done .

Enumerating USBasp HIDUART (Hid Api Library Version : 0.12.0 )
-----------
USBasp List index : 0

Type : 16C0 05DC
Path : \\?\HID#VID_16C0&PID_05DC&MI_01#7&192e7c9b&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Serial number : 1119
Manufacturer  : www.fischl.de
Product       : USBasp
Release       : 1.11
Interface     : 1
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p t88 -U .\entest.hex
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9311 (probably t88)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\entest.hex for flash
         with 512 bytes in 1 section within [0, 0x1ff]
         using 8 pages and 0 pad bytes
avrdude: writing 512 bytes flash ...
Writing | ################################################## | 100% 0.16 s
avrdude: 512 bytes of flash written
avrdude: verifying flash memory against .\entest.hex
Reading | ################################################## | 100% 0.11 s
avrdude: 512 bytes of flash verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p m32a -U .\urboot_atmega32a_autobaud_uart0_rxd0_txd1_ee_lednop_fr_ce_hw.hex
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9502 (probably m32a)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\urboot_atmega32a_autobaud_uart0_rxd0_txd1_ee_lednop_fr_ce_hw.hex for flash
         with 346 bytes in 2 sections within [0x7e00, 0x7fff]
         using 4 pages and 166 pad bytes
avrdude: writing 346 bytes flash ...
Writing | ################################################## | 100% 0.05 s
avrdude: 346 bytes of flash written
avrdude: verifying flash memory against .\urboot_atmega32a_autobaud_uart0_rxd0_txd1_ee_lednop_fr_ce_hw.hex
Reading | ################################################## | 100% 0.01 s
avrdude: 346 bytes of flash verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p at89s52 -U .\entest.hex
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e5206 (probably 89S52)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\entest.hex for flash
         with 512 bytes in 1 section within [0, 0x1ff]
avrdude: writing 512 bytes flash ...
Writing | ################################################## | 100% 1.33 s
avrdude: 512 bytes of flash written
avrdude: verifying flash memory against .\entest.hex
Reading | ################################################## | 100% 0.71 s
avrdude: 512 bytes of flash verified

avrdude done.  Thank you.
```

Franck78, 2023-05-07:
```Writing | ################################################## | 100% 74.42s

avrdude: 260718 bytes of flash written
avrdude: verifying flash memory against Chip-TesterPro-FW-v0.24.hex:
avrdude: load data flash data from input file Chip-TesterPro-FW-v0.24.hex:
avrdude: input file Chip-TesterPro-FW-v0.24.hex contains 260718 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 62.93s

avrdude: verifying ...
avrdude: 260718 bytes of flash verified
```
working

commit a31296e9 (2023-05-07): Revert the second HID interface poll interval to 20ms as the 10ms is too fast . Fixes #23