dioannidis/usbasp #24: USBaspHIDUART -u 1000

Franck78, 2023-05-07:
The serial is not updated on my installation.

Perhaps 'hiblib' too old ?

Can you build the USBaspHIDUART with this lib in your code ?

https://github.com/libusb/hidapi/blob/master/BUILD.md#embedding-hidapi-directly-into-your-source-tree

```
:~/bin> USBaspHIDUART -u 3456

Enumerating USBasp HIDUART (Hid Api Library Version : 0.9.0_or_older )
-----------
USBasp List index : 0

Type : 16C0 05DC
Path : 0002:000d:01
Serial number : ????
Manufacturer  : www.fischl.de
Product       : USBasp
Release       : 1.11
Interface     : 1

Updating USBasp Serial Number ... Done .

Enumerating USBasp HIDUART (Hid Api Library Version : 0.9.0_or_older )
-----------
USBasp List index : 0

Type : 16C0 05DC
Path : 0002:000d:01
Serial number : ????
Manufacturer  : www.fischl.de
Product       : USBasp
Release       : 1.11
Interface     : 1
```

dioannidis, 2023-05-07:
Hi,

  which hidapi version do you have ?
 
Isn't possible to upgrade to a newer version ? 0.9.0 is too old ...

regard,

Franck78, 2023-05-07:
not now for the update of my opensuse.

As USBaspHIDUART may run on strange linux systems, maybe not a bad idea to eliminate a potential problem with embedding libs ?

mcuee, 2023-05-08:
For Linux, usually it is better that the user should build from source, if the provided binary does not work.

Take note for avrdude project, we only provide Windows MSVC binary.
https://github.com/avrdudes/avrdude/releases/tag/v7.1

There is an enhancement request but nobody is working on it. Rather we refer the users to build from source or get static linking binary from Arduino-packing project.
https://github.com/avrdudes/avrdude/issues/987

mcuee, 2023-05-08:
BTW, for Linux and macOS, usually the preferred way is to use dynamic linking.

Franck78, 2023-05-08:
Turbo Pascal 5 was my first language. So nothing strange to me when I read for example the bootloader you are contributing to. Even if I don't see the purpose of it (another boot loader).

Some years ago I dived a little in the the USB protocol (bulk, interrupt, endpoint etc etc).  Most of it vanished.

@mcuee 
yes, compiling from source, if you have time to cross compile when your device have no room for gcc, make, libs, headers, blah blah. And infinite rabbit hole.
I sometime reread for fun this https://groups.google.com/g/techos/c/63PDG_3vNns (in french sorry). Keyword=unikernel

So, for me this lib is/was a perfect candidate to be 'included' (eg eliminated) ;)

dioannidis, 2023-05-08:
Hi,

 @Franck78 

 FYI, If you want to change the serial number you could edit the eeprom.eep.hex ...

```
:0A0000000A03300030003000300029
:00000001FF
```

The serial 0000 is stored as ASCII ( 0 -> 30 ) ( shows bold, italic above).

So if you want to change it to 1234 then edit it to look like these :

```
:0A0000000A03310032003300340029
:00000001FF
```

regards,

Franck78, 2023-05-08:
I don't have have need to change it. I wanted to test only.

What is puzzling me is low speed usb unable to expose as a serial port (/dev/acm). Or not able to transfer payload of more than 8 bytes. I have to re-read all this USB stuff if I want to do something in the usbasp code !

dioannidis, 2023-05-08:
Hi,

  @Franck78

> What is puzzling me is low speed usb unable to expose as a serial port (/dev/acm). Or not able to transfer payload of more than 8 bytes.

 It's simple, USB specs . No bulk or no isochronous transfers, only control and interrupt with max 8 byte payload. 

regards,

dioannidis, 2023-05-08:
> Hi,
> 
> @Franck78
> 
> > What is puzzling me is low speed usb unable to expose as a serial port (/dev/acm). Or not able to transfer payload of more than 8 bytes.
> 
> It's simple, USB specs . No bulk or isochronous transfers, only control and interrupt and max 8 byte payload.
> 
> regards,

mcuee, 2023-05-08:
USB 2.0 spec is actually a good read (The Original USB 2.0 specification released on April 27, 2000). It can be downloaded from here.
https://www.usb.org/document-library/usb-20-specification

As for a quick read, I think USB Made Simple is good.
https://www.usbmadesimple.co.uk/