dioannidis/usbasp #25: Which hex to flash?

dzalf, 2023-05-22:
Hi @dioannidis  and contributors

I would like to ask you an extremely _noobish_ question.

I am about to update the firmware from a generic USBasp based on the ATMEGA8A

According to the Readme.txt file (which seems to be a copy/dragged version from the original file from [Thomas Fischl](https://www.fischl.de/usbasp/)) I should flash the file: 
```
usbasp.atmega8.2011-05-28.hex
```
However, what confuses me is the date from this file: xxxxx.2011-05-28

Has this precompiled version been modified to the latest modifications described in this fork? In other words, does this hex file offer the improvements described here? If so, why not update the file naming convention to the most recent date?

Otherwise, please point me to the right file to be flashed.

Apologies if I am missing something obvious

Thanks

dioannidis, 2023-05-22:
Hi,

  you'll find binaries of this fork at the [1.11 pre release](https://github.com/dioannidis/usbasp/releases/tag/v1.11) page. For atmega8a USBasp device you should use the  Firmware_ATMega8_12MHz_TPI_HIDUART_SNWRITE.zip file.

 Mind you that there is a bug that I couldn't find time to look at it yet . See [Error in read after entering PE with slow clocks](#22) .

regards,

dioannidis, 2023-06-05:
@dzalf 

The [Error in read after entering PE with slow clocks](#22) is resolved, so you can use the firmware files from the  [v1.11 release](https://github.com/dioannidis/usbasp/releases/tag/v1.11) page .

( don't forget to flash also the eeprom hex )

regards

closed 2023-06-05 by dioannidis