dioannidis/usbasp #30: Can't get it working with win10

usr97629238, 2024-01-03:
Works perfect across the table on linux, but no matter what I try it won't work with windows (tried with avrdude [7.0](https://github.com/mcuee/avrdude/releases/tag/v7.0) and [7.2_mingw64.zip](https://github.com/mcuee/avrdude/releases/download/v7.2/avrdude-v7.2_mingw64.zip))

USBaspHIDUART.exe sees it connected:
```
Enumerating USBasp HIDUART (Hid Api Library Version : 0.12.0 )
-----------
USBasp List index : 0

Type : 16C0 05DC
Path : \\?\HID#VID_16C0&PID_05DC&MI_01#8&69f9ab7&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Serial number : 0000
Manufacturer  : www.fischl.de
Product       : USBasp
Release       : 1.11
Interface     : 1

-----------
USBasp List index : 1

Type : 16C0 05DC
Path : \\?\HID#VID_16C0&PID_05DC&MI_02#8&2a76d879&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Serial number : 0000
Manufacturer  : www.fischl.de
Product       : USBasp
Release       : 1.11
Interface     : 2
```

But avrdude fails at avrdude usbasp_transmit():
```
avrdude: Version 7.2
         Using Port                    : USB
         Using Programmer              : usbasp
         Setting bit clk period        : 10.0
avrdude: seen device from vendor >www.fischl.de<
avrdude: seen product >USBasp<
avrdude: input file attiny85.ino.hex auto detected as Intel Hex
         AVR Part                      : ATtiny85
         Chip Erase delay              : 400000 us
         RESET disposition             : possible i/o
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
...

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/
avrdude: usbasp_initialize()
avrdude: usbasp_transmit("USBASP_FUNC_GETCAPABILITIES", 0x00, 0x00, 0x00, 0x00)
avrdude usbasp_transmit() [usbasp.c:365] OS error: No such file or directory
avrdude: usbasp_spi_set_sck_period(5e-06)
avrdude: try to set SCK period to 5e-06 s (= 200000 Hz)
avrdude: connected USBasp is not cabable of 3 MHz SCK
avrdude: set SCK frequency to 187500 Hz
avrdude: usbasp_transmit("USBASP_FUNC_SETISPSCK", 0x09, 0x00, 0x00, 0x00)
avrdude usbasp_transmit() [usbasp.c:365] OS error: No such file or directory
avrdude usbasp_spi_set_sck_period() [usbasp.c:979] error: cannot set sck period; please check for usbasp firmware update
avrdude: usbasp_transmit("USBASP_FUNC_CONNECT", 0x00, 0x00, 0x00, 0x00)
avrdude usbasp_transmit() [usbasp.c:365] OS error: No such file or directory
avrdude: usbasp_program_enable()
avrdude: usbasp_transmit("USBASP_FUNC_ENABLEPROG", 0x00, 0x00, 0x00, 0x00)
avrdude usbasp_transmit() [usbasp.c:365] OS error: No such file or directory
avrdude usbasp_spi_program_enable() [usbasp.c:734] error: program enable: target does not answer (0x00)
avrdude main() [main.c:1301] error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower the bit clock frequency, e.g. -B 125kHz
        - use -F to override this check
avrdude: usbasp_close()
avrdude: usbasp_transmit("USBASP_FUNC_DISCONNECT", 0x00, 0x00, 0x00, 0x00)
avrdude usbasp_transmit() [usbasp.c:365] OS error: No such file or directory
avrdude done.  Thank you.
```
Changing the -B parameter affects nothing.

dioannidis, 2024-01-04:
Hi @usr97629238 ,

Is the USBasp's device driver WINUSB ? You can check using the device manager ( the image at https://github.com/dioannidis/usbasp/discussions/29 shows how the USBasp using WINUSB looks like ) .

Also, maybe there are driver conflicts from previous libusb and/or zadig installations . Use device manager to uninstall / delete all the drivers that Windows associates to USBasp device . Using device manager, right click on the usbasp device and uninstall / delete all the drivers assigned to it until windows reports it as unknown . Then unplugged it and plugged it again. It should work .

regards,

usr97629238, 2024-01-04:
Thanks! Uninstalling the drivers somehow fixed it, despite the fact that there weren't any libusb and/or zadig drivers ever installed.

usr97629238, 2024-01-05:
Actually there is more to the issue:

- It turns out it only works until the next disconnect, then (on the next reconnect) it would require another round of uninstall the driver, reconnect the dongle. That's on ``USB 2.0 ports that are Intel(R) 6 Series/C200 Series Chipset Family USB Enhanced Host Controller (PCI\VEN_8086&DEV_1C2D&SUBSYS_844D1043&REV_05)``
- It won't autoinstall the winusb.sys driver on the ``ASMedia USB 3.0 controller (PCI\VEN_1B21&DEV_1042&SUBSYS_84881043&REV_00)`` and show ``"There are no compatible drivers for this device"``, if I manually install the winusb.sys it would accept it but fail the same at avrdude usbasp_transmit()
All these use WHQL drivers autoinstalled by windows. And actually there aren't any newer drivers anyway.

dioannidis, 2024-01-06:
Hi,

  maybe you should clean the Windows Registry reg. USBasp . 

  This is how I do it using [USBDview](http://www.nirsoft.net/utils/usb_devices_view.html) . **Run USBDView as Administrator**, go to Options and check the "Display Disconnected Devices" . Sort the list clicking at the VendorID Header, select all the devices with VendorID = 16c0 and ProductID = 05dc. Right-Click on one of the selected device and uninstall them using the "Uninstall Selected Devices" menu item .

i.e. :

[image]

Hope it helps .

regards,

usr97629238, 2024-01-07:
[image]

Nah, like I've said, there were no other drivers, and I'm usually wary with spamming the system with hacks like zadig and questionable drivers.

dioannidis, 2024-01-07:
Hi,

 did you uninstalled those devices I see in your post, tried again and you get the same results ?

regards,

usr97629238, 2024-01-08:
Yes, removed all the 5 drivers and did a fresh re-test and here how it goes:
- It won't install the winusb.sys driver on the ASMedia controller, and it won't work after that even on the Intel controller (the driver shows up installed, however the flashing fails with the ``usbasp_transmit() [usbasp.c:365] OS error: No such file or directory``) unless I do the "uninstall the driver, reconnect the dongle"

[image]

- It works fine on the Intel controller

And I now believe that it wasn't working (the first post) because I've inserted it into the ASMedia controller first.

So it's up to you yo decide if you want to investigate this any further or just close the issue. My take is that it's due to a flaky ASMedia controller or driver. If you want me to perform any tests of what might be wrong with it - feel free to ask.

closed 2025-02-22 by dioannidis