dioannidis/usbasp #33: AVRDUDE 8.0 can't see the programmer under Windows 11

TzOk83, 2025-02-21:
I can't make USBasp running your improved v1.11 firmware to work under Windows 11.
I keep getting errors like this:
`>>>: avrdude.exe -c usbasp-clone -P usb -p m8 
WARNING: Unable to detect MCU

Error: cannot find USB device with vid=0x16c0 pid=0x5dc
Error: unable to open port usb for programmer usbasp-clone

Avrdude done.  Thank you.`

The device is detected as USBasp with ID USB\VID_16C0&PID_05DC&MI_00\9&139AF701&0&0000.

Zadig sees USBasp (interface 0) as WINUSB (v10.0.26100.1150) and USBasp (interface 1) and USBasp (Interface 2), both as HidUsb (v10.0.26100.1882) devices. I haven't tried to install any other drivers with Zadig.

ccrause, 2025-02-22:
Perhaps related to [this note](https://github.com/dioannidis/usbasp?tab=readme-ov-file#avrdude) about avrdude?  You mention avrdude 8.0, so I assume it is the official release and not the unofficial version with libusb from @mcuee.

dioannidis, 2025-02-22:
Hi,

  as @ccrause wrote, you have to use a msys2 / mingw avrdude build on windows for this firmware . It's also noted in the avrdude 8.0 release notes ( https://github.com/avrdudes/avrdude/releases/tag/v8.0 ) . 

 Please try the https://github.com/avrdudes/avrdude/releases/download/v8.0/avrdude-v8.0-windows-mingw-x64.zip .

regards,

TzOk83, 2025-02-22:
I tried the @mcuee release but without success. Apparently, I have chosen the wrong version, as I now read that "avrdude-v7.3-windows-*.zip -- Same binaries as official Avrdude GitHub page. These binaries will print Avrdude version 7.3."

I have also read the Avrdude 8.0 release notes, and indeed there is a note "not working with USB Composite Device programmer like USBASP with the FW from https://github.com/dioannidis/usbasp", but there is no clear information on which version to use instead. On the other hand, I remembered using AVRDUDESS with Avrdude 8.0 with USBasp V1.11 (Dioannidis) firmware on my old computer.

Maybe it would be worth adding some instructions on how to use this firmware with Arduino and Avrdude.

dioannidis, 2025-02-22:
Hi,

@TzOk83 

  I assume, from your response, that with the MSYS2 mingw avrdude build your USBasp is working now, correct ?

> Maybe it would be worth adding some instructions on how to use this firmware with Arduino and Avrdude.

  You mean in the USBasp README ? 
  
regards,

mcuee, 2025-02-22:
Hmm, I think avrdude 8.0 github binary download link is clear enough. I wrote those notes.

https://github.com/avrdudes/avrdude/releases/tag/v8.0

Preferred Windows binaries to use:
avrdude-v8.0-windows-x64.zip --> Windows MSVC 64 bit
avrdude-v8.0-windows-mingw-x64.zip --> Windows MSYS2 mingw 64 bit

Static build binaries (*.tar.gz) are recommended for Arduino users. Take note they do not contain libserialport support due to the limitation of the crossbuild container.

**Notes for the avrdude Windows binaries (static linked to the dependency libraries) :**

1) MSVC build is recommended for most Windows users unless you hit known limitations (see below). The majority of users should download `avrdude-v8.0-windows-x64.zip`. In case one of limitations below is unacceptable you may want to download `avrdude-v8.0-windows_mingw-x64.zip`.

2) Known limitation for the above MSVC build (`avrdude-v8.0-windows-x64.zip`, `avrdude-v8.0-windows-x86.zip` and `avrdude-v8.0-windows-arm64.zip`) 

- https://github.com/avrdudes/avrdude/issues/1552 (no libserialport support)
- https://github.com/avrdudes/avrdude/issues/1253 (slow speed when using arduino-ft232r or similar programmer)
- https://github.com/avrdudes/avrdude/issues/968 (not working with USB Composite Device programmer like USBASP with the FW from https://github.com/dioannidis/usbasp)
- https://github.com/avrdudes/avrdude/discussions/1440 (no support of CH341A)

3) Known limitation for the MSYS2 mingw build (`avrdude-v8.0-windows_mingw-x64.zip`, `avrdude-v8.0-windows_mingw-clang-x64.zip` and `avrdude-v8.0-windows_mingw-ucrt-x64.zip`)

- You will have to use Zadig to switch the FTDI vendor driver if you use libftdi based programmer (eg: ft232h, ft2232h, arduino-ft232r)
- https://github.com/avrdudes/avrdude/issues/1271 (timeout may be more likely to happen for urclock in terminal mode)

TzOk83, 2025-02-22:
Basically, I gave up and concluded that Ralph Doncaster's build, although messy (go, find the right file to flash in his repo), was more suited for me. I don't have to remember to change the avrdude executable in Arduino and Avrdudess after every update nor mess around with Zadig.
You did a lot of good job with this firmware, but I'm not sure if it should be still called USBasp, as it is not compatible with the original, unmodified software for this programmer.

P.S.
I'd put a big message in readme - works only with MSYS2 mingw builds of Avrdude (because it works with the official build, but not the default one, now it is even not about official vs. @mcuee but MSVC vs MSYS2).

closed 2025-02-22 by TzOk83